osrf/capabilities #86: Not able to define roscpp services in capability package.

primohan01, 2017-06-28:
After running the capability server i am getting below warnings. No defualt provider given for capability interface minimal_pkg/slowtodie and minimal_pkg/specific minimal has only one provider.

I have services written in roscpp and want to define those services in capability server package so that those services can be controlled from this package. but i am newbie to python/yaml so not sure where this services needs to be define. in package.xml file all the capability server related services are defined. kindly let me know the approach so that i can explore from my end. Currently i am able to up the server but not sure how to define my services through capability that part i am not aware.

primohan01, 2017-07-02:
Can anybody throw some light on this or share some tutorial so that i can proceed further.i will appreciate if any online tutorial link is shared. It looks like most of the people are able to start and stop their service through registering service by capability server. I am able to up the capability server but not sure what information needs to be updated in yaml file.

wjwwood, 2017-07-17:
This page is linked from the `README.md`: http://wiki.ros.org/capabilities/Tutorials

This doesn't sound like a bug, so I'd suggest you ask on answers.ros.org to see if anyone there can help you. I'll close this until we find some issue with this package.

closed 2017-07-17 by wjwwood